rsasaki0109/li_slam_ros2 #1: Compile prompt error

hunkyu, 2020-10-07:
scanmatcher prompt:
<command-line>:0:1: error: macro names must be identifiers

Found in image_projection.cpp 
`add_executable(image_projection src/image_projection.cpp)
  ament_target_dependencies(image_projection
    rclcpp
    rclcpp_components
    tf2_ros
    tf2_geometry_msgs
    tf2_sensor_msgs
    tf2_eigen
    geometry_msgs
    sensor_msgs
    nav_msgs
    #liosam_msgs
    PCL
    OpenCV
  )`
I do not compile and generate image_projection, but there is no problem

rsasaki0109, 2020-10-08:
Hmmm, I've never faced that error before, so I don't know how to deal with it

hunkyu, 2020-10-08:
Which version of ros2 are you currently using? ros2 dashing?

rsasaki0109, 2020-10-08:
ros2 foxy

hunkyu, 2020-10-10:
Finished <<< ndt_omp_ros2 [38.6s]
Starting >>> graph_based_slam
Starting >>> scanmatcher                        
[Processing: graph_based_slam, scanmatcher]                                                      
[Processing: graph_based_slam, scanmatcher]                                         
--- stderr: scanmatcher                                                             
** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to ensenso will be disabled
** WARNING ** visualization features related to davidSDK will be disabled
** WARNING ** visualization features related to dssdk will be disabled
** WARNING ** visualization features related to rssdk will be disabled
<command-line>:0:1: error: macro names must be identifiers
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/image_projection.cpp:169:65: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
   void imuHandler(const sensor_msgs::msg::Imu::ConstPtr & imuMsg)
                                                                 ^
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/image_projection.cpp:194:77: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
   void odometryHandler(const nav_msgs::msg::Odometry::ConstPtr & odometryMsg)
                                                                             ^
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/image_projection.cpp:200:82: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
   void cloudHandler(const sensor_msgs::msg::PointCloud2::ConstPtr & laserCloudMsg)
                                                                                  ^
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/image_projection.cpp:219:85: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
   bool cachePointCloud(const sensor_msgs::msg::PointCloud2::ConstPtr & laserCloudMsg)
                                                                                     ^
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/image_projection.cpp: In constructor ‘ImageProjection::ImageProjection(const rclcpp::NodeOptions&)’:
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/image_projection.cpp:109:60: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
       [this](const sensor_msgs::msg::Imu::ConstPtr msg) -> void
                                                            ^~~~
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/image_projection.cpp:116:62: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
       [this](const nav_msgs::msg::Odometry::ConstPtr msg) -> void
                                                              ^~~~
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/image_projection.cpp:123:68: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
       [this](const sensor_msgs::msg::PointCloud2::ConstPtr msg) -> void
                                                                    ^~~~
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/imu_preintegration.cpp:196:73: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
   void odometryHandler(const nav_msgs::msg::Odometry::ConstPtr & odomMsg)
                                                                         ^
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/imu_preintegration.cpp:415:66: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
   void imuHandler(const sensor_msgs::msg::Imu::ConstPtr & imu_raw)
                                                                  ^
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/imu_preintegration.cpp: In constructor ‘IMUPreintegration::IMUPreintegration(const rclcpp::NodeOptions&)’:
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/imu_preintegration.cpp:125:60: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
       [this](const sensor_msgs::msg::Imu::ConstPtr msg) -> void
                                                            ^~~~
/home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/imu_preintegration.cpp:130:62: warning: ‘ConstPtr’ is deprecated [-Wdeprecated-declarations]
       [this](const nav_msgs::msg::Odometry::ConstPtr msg) -> void
                                                              ^~~~
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52:0,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
                 from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
                 from /home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/include/scanmatcher/utility.h:55,
                 from /home/gzxsmapper/ros2_lio_ws/src/li_slam_ros2-master/scanmatcher/src/image_projection.cpp:31:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
[... cut ...]
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
make: *** [all] Error 2
---
Failed   <<< scanmatcher [1min 6s, exited with code 2]

hunkyu, 2020-10-10:
My usage time ubuntu 18.04 ros-dasing

hunkyu, 2020-10-10:
[image]

rsasaki0109, 2020-10-10:
It looks like you wanted to upload an image, but it's not showing up and the link is weird.

hunkyu, 2020-10-10:
[image]
The trajectory has drifted, which may be a problem with rosbag. I am using ros2 run ros1_bridge dynamic_bridge --bridge-all-topics

rsasaki0109, 2020-10-10:
Details, please.

JakeReeves21, 2021-01-13:
Hello, I have been trying to get ros2 to work with this solution for a few days. I have gotten the rviz2 and ros2 launch scanmatcher commands to run correctly in separate windows.

I have not been able to get the "ros2 bag play -s rosbag_v2 casual_walk.bag" to run correctly in its own window. It cannot read the Velodyne topics, because those topics seem to have been recorded in with ROS 1. Please see below screenshot.

Any ideas on what I am doing incorrectly?

[image]

closed 2021-03-05 by rsasaki0109

Gaoeee, 2023-08-14:
> scanmatcher prompt: :0:1: error: macro names must be identifiers
> 
> Found in image_projection.cpp `add_executable(image_projection src/image_projection.cpp) ament_target_dependencies(image_projection rclcpp rclcpp_components tf2_ros tf2_geometry_msgs tf2_sensor_msgs tf2_eigen geometry_msgs sensor_msgs nav_msgs #liosam_msgs PCL OpenCV )` I do not compile and generate image_projection, but there is no problem

how did you solve it?