rsasaki0109/li_slam_ros2 #2: Error occurs when launching the lio.launch.py.

CAKGOD, 2020-11-09:
Hello. Bother you again.
I build the li_slam_ros2 successfully but there is an error when I use the command "ros2 launch scanmatcher lio.launch.py". The detailed info is shown below. Can you help me solve it? Thank you!
`[INFO] [launch]: All log files can be found below /home/cakgod/.ros/log/2020-11-09-03-25-29-689650-cakgod-ThinkCentre-M910x-N000-75592
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=SubstitutionFailure("executable 'imu_preintegration' not found on the libexec directory '/home/cakgod/ros2_ws/install/scanmatcher/lib/scanmatcher' ")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 442, in execute
    ret = super().execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 775, in execute
    self.__expand_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 635, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 635, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 184, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/substitutions/executable_in_package.py", line 84, in perform
    raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: executable 'imu_preintegration' not found on the libexec directory '/home/cakgod/ros2_ws/install/scanmatcher/lib/scanmatcher' 
Task exception was never retrieved
future: <Task finished name='Task-12' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 421, in __on_signal_process_event
    raise RuntimeError('Signal event received before subprocess transport available.')
RuntimeError: Signal event received before subprocess transport available.
[INFO] [scanmatcher_node-1]: process started with pid [75594]
[INFO] [static_transform_publisher-2]: process started with pid [75596]
[INFO] [scanmatcher_node-1]: sending signal 'SIGINT' to process[scanmatcher_node-1]
[ERROR] [scanmatcher_node-1]: process has died [pid 75594, exit code -2, cmd '/home/cakgod/ros2_ws/install/scanmatcher/lib/scanmatcher/scanmatcher_node --ros-args -r /input_cloud:=/cloud_deskewed --params-file /home/cakgod/ros2_ws/install/scanmatcher/share/scanmatcher/param/lio.yaml'].
[static_transform_publisher-2] [INFO] [1604910329.767936466] [static_transform_publisher_I3XGArIQqPVe3muV]: Spinning until killed publishing transform from 'base_link' to 'velodyne'
[ERROR] [static_transform_publisher-2]: process[static_transform_publisher-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [static_transform_publisher-2]: sending signal 'SIGTERM' to process[static_transform_publisher-2]
[ERROR] [static_transform_publisher-2]: process has died [pid 75596, exit code -15, cmd '/opt/ros/foxy/lib/tf2_ros/static_transform_publisher 0 0 0 0 0 0 1 base_link velodyne --ros-args'].
`

CAKGOD, 2020-11-09:
Btw, I use the ubuntu 20.04 and ros2 foxy.

rsasaki0109, 2020-11-09:
Maybe you don't have gtsam installed or linked to li_slam_ros2.

Make sure you can build this package by commenting out 「if(GTSAM_FOUND)」 and 「endif()」 below.
https://github.com/rsasaki0109/li_slam_ros2/blob/master/scanmatcher/CMakeLists.txt#L102-L105
https://github.com/rsasaki0109/li_slam_ros2/blob/master/scanmatcher/CMakeLists.txt#L134-L179

CAKGOD, 2020-11-13:
Thanks. It works after reinstalling the GTSAM.
There is another question. Did this framework support other lidar datasets well? For example, the 16/32/64 layers dataset. Have you test it on such datasets?
Looking forward to hearing from you.

rsasaki0109, 2020-11-13:
I think this package would work with other datasets as well, but I haven't tried it. I think it works, at least with the data described in the LIO-SAM README.md.

CAKGOD, 2020-11-16:
Hi! Thank you for your great work!
I was working on looking for a good ros2-based slam frameworks and testing
them. Some ros1-based slam frameworks including hdl_graph_slam, lego_loam,
and lio-sam had been test. You can find the test report in the project
slam_test <https://github.com/CAKGOD/slam_test>. Some ros2-based slam
frameworks including lidarslam_ros2 and li_slam_ros2 were also test.
(Without any offense, I found that it was harder in parameter adjustments
about your framework. I believe it is because of my insufficient ability.)
I need to select a ros2-based slam framework from them to contribute to
ros2 community. Of course, there is a way for me to do this. Now there are
two ways to complete it. One is porting the lio-sam to ros2 and the other
is contributing the li_slam_ros2 (if it is agreed). But I found some
matters in the porting process of lio-sam such as the unporting of pcl_ros.
So we can contribute the li_slam_ros2 if it has better slam ability and you
agree with it of course.
So I want to know your opinion on this.

rsasaki0109, 2020-11-16:
Maybe the essential problem is not parameter tuning.
The reason it doesn't work with the KITTI you're testing is because I developed this package for a mobile robot that uses a 16-line lidar.

The accuracy of my package is equal to or better than other SLAM packages (I think NDT / GICP is considerably more accurate than LOAM). The problem, however, is that the processing is heavy.

The package I am using is not very suitable for cars. For car scan matching, it is very important to estimate the initial value from the previous position and the previous two positions, but this feature is not implemented in this package because it is not important for mobile robots. Without this feature, the NDT calculation time is even longer and real-time scan matching in the vehicle is more difficult

So I find it difficult to process KITTI data in real time.
Maybe slowing down the Rosbag would make it a little better.

CAKGOD, 2020-11-19:
Thank you for your reply.

closed 2020-11-19 by CAKGOD