rsasaki0109/li_slam_ros2 #3: is the git command in readme correct ?

jediofgever, 2020-12-16:
Hi , 
I was trying out this package but there is some issue I guess. 
Inn readme you never mention a git command  to clone this repo but instead another repo(again your repo), is this intentional ? 

Could you please check and confirm ? 
Thanks for the awesome work ,

rsasaki0109, 2020-12-16:
My mistake. Thank you for pointing it out.

jediofgever, 2020-12-16:
No worries, awesome work already. I am trying to test it in simulation, have had a chance to test this under Gazebo or any other simulation env. ?

rsasaki0109, 2020-12-16:
Unfortunately, I don't have a good enough PC to simulate lidar, so I haven't tried it.

jediofgever, 2020-12-16:
I actually tried it , but since I am using navigation, there is quite a lot of message topic confliction. I solvbed some of them but I cant achieve the mapping. The map is constantly shifting.

I am not sure what I am doing wrong here but I guess this could be related to wrong odometry ? , Do you have any suggestions ? 

[image]

rsasaki0109, 2020-12-17:
Check if self-positioning can be estimated by mapping using only Lidar.

jediofgever, 2020-12-18:
unfortunately with perfectly fine simulated lidar data I cant create maps

closed 2020-12-18 by jediofgever