rsasaki0109/li_slam_ros2 #4: ros2 bag play issue

JakeReeves21, 2021-01-13:
Hello, I have been trying to get ros2 to work with this solution for a few days. I have gotten the rviz2 and ros2 launch scanmatcher commands to run correctly in separate windows.

I have not been able to get the "ros2 bag play -s rosbag_v2 casual_walk.bag" to run correctly in its own window. It cannot read the Velodyne topics, because those topics seem to have been recorded in with ROS 1. Please see below screenshot.

Any ideas on what I am doing incorrectly?

[image]

rsasaki0109, 2021-01-13:
The /velodyne_* topics are not subscribed to by li_slam_ros2.
li_slam_ros2 will subscribe to /points_raw and /imu_raw or /imu_correct.

JakeReeves21, 2021-01-13:
So after playing back the casual_walk.bag, I should be seeing the path visualized in rviz2, correct?

rsasaki0109, 2021-01-13:
Yes.
Did you use 「ros2 topic echo」 to see if the path or odom topic is output?

JakeReeves21, 2021-01-13:
Okay, I rebooted my system and I am now getting the jagged output that the other issue was having. I will try to slow down the bag play or use more threads as you suggested in the other issue before I close this issue.

JakeReeves21, 2021-01-13:
Seems to be working by slowing the bag playback down. How do we make scanmatcher use more threads? Do we increase the -j4 number when making it?

JakeReeves21, 2021-01-13:
The path is very clear on the casual_walk.bag playback. However, I am not see any buildings in the lidar scan. How were you able to display the lidar scans in your examples if li_slam_ros2 does not subscribe to the velodyne topics?

rsasaki0109, 2021-01-13:
Adjust the ndt_num_threads value below (0 means all threads are used).
https://github.com/rsasaki0109/li_slam_ros2/blob/master/scanmatcher/param/lio.yaml#L7

If you use too many threads, it may overload your PC and cause scan matching to fail, so you need to choose a value that is just right.

rsasaki0109, 2021-01-13:
> How were you able to display the lidar scans in your examples if li_slam_ros2 does not subscribe to the velodyne topics?
The map becomes visible when you add a topic in rviz.
Since li_slam_ros2 publishes self-positioned tf, I think scan will also become visible after adding a topic.

JakeReeves21, 2021-01-13:
Sounds great. Thanks for your help! I have been playing around with rviz for a few minutes, and I have gotten it to show a portion of the pointcloud.

I will try scan next.

closed 2021-01-13 by JakeReeves21