rsasaki0109/li_slam_ros2 #6: OS1-32 Lidar compatibility?

MrOCW, 2022-07-06:
Hi, using an external 9DoF IMU with an Ouster1-32,
I am playing from a rosbag but im getting such PCL errors.
```
[image_projection-3] [WARN] [1657123061.814967364] [image_projection]: Point cloud timestamp not available, deskew function disabled, system will drift significantly!
[image_projection-3] Failed to find match for field 'ring'.
[image_projection-3] Failed to find match for field 'time'.
```

rsasaki0109, 2022-07-06:
This package is currently not ouster-compatible.
I will try to support it this weekend if I can.

rsasaki0109, 2022-07-11:
I just need to implement the following, but I was too busy last weekend to do so. I will commit when I have time.

```
        else if (sensor == SensorType::OUSTER)
        {
            // Convert to Velodyne format
            pcl::moveFromROSMsg(currentCloudMsg, *tmpOusterCloudIn);
            laserCloudIn->points.resize(tmpOusterCloudIn->size());
            laserCloudIn->is_dense = tmpOusterCloudIn->is_dense;
            for (size_t i = 0; i < tmpOusterCloudIn->size(); i++)
            {
                auto &src = tmpOusterCloudIn->points[i];
                auto &dst = laserCloudIn->points[i];
                dst.x = src.x;
                dst.y = src.y;
                dst.z = src.z;
                dst.intensity = src.intensity;
                dst.ring = src.ring;
                dst.time = src.t * 1e-9f;
            }
        }
```
https://github.com/TixiaoShan/LIO-SAM/blob/master/src/imageProjection.cpp#L211-L228

MrOCW, 2022-07-12:
Hi @rsasaki0109, thanks so much but take your time no hurry! I managed to get a 3d map using LIO-SAM already

rsasaki0109, 2022-07-13:
Ok, I think it is important to respond to ouster, so I will not close this issue yet, but will close this issue when it has been addressed.
Thank you for putting up the issue.

rsasaki0109, 2022-07-29:
I close this issue as I have merged the PR for ouster

Support ouster and livox 
https://github.com/rsasaki0109/li_slam_ros2/pull/7

closed 2022-07-29 by rsasaki0109