rsasaki0109/li_slam_ros2 #9: li_slam won't start mapping

SyahirMuzni, 2022-09-29:
Hi Mr. Sasaki.

I tried to launch li_slam. But it won't start to create map. Are there any steps that I missed? 

[image]

rsasaki0109, 2022-09-29:
How is it on rviz?
Have you configured imu properly? If not properly, LiDAR Inertial SLAM will not work.
https://github.com/rsasaki0109/li_slam_ros2/blob/develop/scanmatcher/param/lio.yaml#L9-L20
I would like to know information about IMU.

・what kind of IMU you are using.
(an IMU of about $10 may not work well)
・Frequency
tf of LiDAR-IMU
・Noise and bias of IMU
If you have not estimated the noise or bias of the IMU, please use the following package to do so.
https://github.com/gaowenliang/imu_utils

Translated with www.DeepL.com/Translator (free version)

SyahirMuzni, 2022-10-13:
Hi there, 
Ok, I am currently working on the IMU configuration (noise & bias). For your information, I am using IMU (Xsens MTI-G-710). But how do I get the extrinsic Trans, Rot & RPY?

rsasaki0109, 2022-10-13:
Please refer to the following
https://github.com/TixiaoShan/LIO-SAM#prepare-imu-data

space192, 2023-06-25:
hey @rsasaki0109 would you recommend any 6-axis IMU to use with this project ?

rsasaki0109, 2023-06-26:
This package is based on LIO-SAM's fusion method, so the following and LIO-SAM's issues may be helpful.

> We use Microstrain 3DM-GX5-25, which outputs data at 500Hz. We recommend using an IMU that gives at least a 200Hz output rate. 
https://github.com/TixiaoShan/LIO-SAM#prepare-imu-data

rsasaki0109, 2023-07-23:
There has been no recent activity on this issue, so I am going to close it.

closed 2023-07-23 by rsasaki0109