rsasaki0109/li_slam_ros2 #13: Ouster Compatability

jeychandar, 2023-01-11:
@rsasaki0109 Hi Sir. I am Working on this repo and it is working perfectly. I have a doubt, does this repo supports ouster internal IMU. Can you please clarify me. Thanks in advance.

rsasaki0109, 2023-01-11:
I have followed LIO-SAM's lead, and I've supported ouster, but I haven't confirmed that it works.

https://github.com/rsasaki0109/li_slam_ros2/blob/develop/scanmatcher/param/lio.yaml#L4-L6
PR
https://github.com/rsasaki0109/li_slam_ros2/pull/7

jeychandar, 2023-01-23:
@rsasaki0109 Hi. I converted ouster pcap to rosbag and then converted that rosbag into ros2 sqlite3 format (.db3 extension). Does our repo supports sqlite3. I have tried but no mapping no initialization nothing happened. Could you please help me out or can you tell me how to run pcap files directly using our repo. This will be helpful Sir.

rsasaki0109, 2023-01-23:
You may refer to the section on ouster below.
https://github.com/TixiaoShan/LIO-SAM#other-notes

jeychandar, 2023-01-24:
Yes Sir I have followed the instruction. Actually I converted pcap format file into bag format using this tool https://github.com/Krishtof-Korda/pcap-to-bag.git. After that I used rosbag info it shows topic info but when i use ros2 bag it doesn't showing the topic info.
[image]

rsasaki0109, 2023-01-24:
It would be better to convert ros1 rosbag to ros2 rosbag with the following package.

https://pypi.org/project/rosbags/

jeychandar, 2023-01-25:
Hi. I converted rosbag to ros2 working format version for rosbag_v2 but I have internal IMU recorded data and I tried to play the bag file.Scanmatcher shows the error. I tried to check in image projection.cpp but i couldnot get rid of the error.

praveen@ubuntu:~/ros2_ws$ ros2 launch scanmatcher lio_ouster.launch.py
[INFO] [launch]: All log files can be found below /home/praveen/.ros/log/2023-01-25-16-45-11-188492-ubuntu-17135
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [scanmatcher_node-1]: process started with pid [17137]
[INFO] [static_transform_publisher-2]: process started with pid [17139]
[INFO] [imu_preintegration-3]: process started with pid [17141]
[INFO] [image_projection-4]: process started with pid [17161]
[INFO] [graph_based_slam_node-5]: process started with pid [17174]
[imu_preintegration-3] ----> IMU Preintegration Started.
[static_transform_publisher-2] [INFO] [1674636311.266374576] [static_transform_publisher_bnSGUpiqTIQPmcAH]: Spinning until killed publishing transform from 'base_link' to 'velodyne'
[scanmatcher_node-1] registration_method:NDT
[scanmatcher_node-1] ndt_resolution[m]:2
[scanmatcher_node-1] ndt_num_threads:0
[scanmatcher_node-1] gicp_corr_dist_threshold[m]:5
[scanmatcher_node-1] trans_for_mapupdate[m]:1.5
[scanmatcher_node-1] vg_size_for_input[m]:0.2
[scanmatcher_node-1] vg_size_for_map[m]:0.2
[scanmatcher_node-1] use_min_max_filter:false
[scanmatcher_node-1] scan_min_range[m]:0.1
[scanmatcher_node-1] scan_max_range[m]:100
[scanmatcher_node-1] set_initial_pose:true
[scanmatcher_node-1] use_odom:false
[scanmatcher_node-1] use_imu:true
[scanmatcher_node-1] debug_flag:false
[scanmatcher_node-1] map_publish_period[sec]:100
[scanmatcher_node-1] num_targeted_cloud:20
[scanmatcher_node-1] ------------------
[scanmatcher_node-1] [INFO] [1674636311.289797348] [scan_matcher]: initialize Publishers and Subscribers
[scanmatcher_node-1] [INFO] [1674636311.296548136] [scan_matcher]: initialization end
[image_projection-4] ----> ImageProjection Started.
[graph_based_slam_node-5] [INFO] [1674636311.353489260] [graph_based_slam]: initialization start
[graph_based_slam_node-5] registration_method:GICP
[graph_based_slam_node-5] voxel_leaf_size[m]:0.2
[graph_based_slam_node-5] ndt_resolution[m]:5
[graph_based_slam_node-5] ndt_num_threads:1
[graph_based_slam_node-5] loop_detection_period[Hz]:3000
[graph_based_slam_node-5] threshold_loop_closure_score:1
[graph_based_slam_node-5] distance_loop_closure[m]:25
[graph_based_slam_node-5] range_of_searching_loop_closure[m]:20
[graph_based_slam_node-5] search_submap_num:4
[graph_based_slam_node-5] num_adjacent_pose_cnstraints:5
[graph_based_slam_node-5] use_save_map_in_loop:false
[graph_based_slam_node-5] ------------------
[graph_based_slam_node-5] [INFO] [1674636311.358652508] [graph_based_slam]: initialize Publishers and Subscribers
[graph_based_slam_node-5] [INFO] [1674636311.361475503] [graph_based_slam]: initialization end
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[scanmatcher_node-1] [INFO] [1674636329.199597075] [scan_matcher]: create a first map
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.

jeychandar, 2023-01-27:
@rsasaki0109 I managed to match the ring but hard to manage noise. Internal IMU data works with scanmatcher and started mapping but I can't solve this **failed to find match noise issue**. I reduced playback speed also but still getting **Large Velocity Reset IMU Preintegration**. I think it is because of noise in image projection. Could you please help me out.

praveen@ubuntu:~/ros2_ws$ ros2 launch scanmatcher lio_ouster_32.launch.py
[INFO] [launch]: All log files can be found below /home/praveen/.ros/log/2023-01-27-14-57-08-996706-ubuntu-23485
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [scanmatcher_node-1]: process started with pid [23487]
[INFO] [static_transform_publisher-2]: process started with pid [23489]
[INFO] [imu_preintegration-3]: process started with pid [23491]
[INFO] [image_projection-4]: process started with pid [23493]
[INFO] [graph_based_slam_node-5]: process started with pid [23499]
[imu_preintegration-3] ----> IMU Preintegration Started.
[static_transform_publisher-2] [INFO] [1674802629.224041429] [static_transform_publisher_MDU8TaMW2eviw9Zq]: Spinning until killed publishing transform from 'base_link' to 'velodyne'
[image_projection-4] ----> ImageProjection Started.
[scanmatcher_node-1] registration_method:NDT
[scanmatcher_node-1] ndt_resolution[m]:2
[scanmatcher_node-1] ndt_num_threads:0
[scanmatcher_node-1] gicp_corr_dist_threshold[m]:5
[scanmatcher_node-1] trans_for_mapupdate[m]:1.5
[scanmatcher_node-1] vg_size_for_input[m]:0.2
[scanmatcher_node-1] vg_size_for_map[m]:0.2
[scanmatcher_node-1] use_min_max_filter:false
[scanmatcher_node-1] scan_min_range[m]:0.1
[scanmatcher_node-1] scan_max_range[m]:100
[scanmatcher_node-1] set_initial_pose:true
[scanmatcher_node-1] use_odom:false
[scanmatcher_node-1] use_imu:true
[scanmatcher_node-1] debug_flag:false
[scanmatcher_node-1] map_publish_period[sec]:100
[scanmatcher_node-1] num_targeted_cloud:20
[scanmatcher_node-1] ------------------
[scanmatcher_node-1] [INFO] [1674802629.274256107] [scan_matcher]: initialize Publishers and Subscribers
[scanmatcher_node-1] [INFO] [1674802629.285926016] [scan_matcher]: initialization end
[graph_based_slam_node-5] [INFO] [1674802629.404393715] [graph_based_slam]: initialization start
[graph_based_slam_node-5] registration_method:GICP
[graph_based_slam_node-5] voxel_leaf_size[m]:0.1
[graph_based_slam_node-5] ndt_resolution[m]:5
[graph_based_slam_node-5] ndt_num_threads:1
[graph_based_slam_node-5] loop_detection_period[Hz]:3000
[graph_based_slam_node-5] threshold_loop_closure_score:1
[graph_based_slam_node-5] distance_loop_closure[m]:25
[graph_based_slam_node-5] range_of_searching_loop_closure[m]:20
[graph_based_slam_node-5] search_submap_num:4
[graph_based_slam_node-5] num_adjacent_pose_cnstraints:5
[graph_based_slam_node-5] use_save_map_in_loop:false
[graph_based_slam_node-5] ------------------
[graph_based_slam_node-5] [INFO] [1674802629.405143711] [graph_based_slam]: initialize Publishers and Subscribers
[graph_based_slam_node-5] [INFO] [1674802629.411520513] [graph_based_slam]: initialization end
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[scanmatcher_node-1] [INFO] [1674802638.088709724] [scan_matcher]: create a first map
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[imu_preintegration-3] [WARN] [1674802646.148734015] [imu_preintegration]: Large velocity, reset IMU-preintegration!
[image_projection-4] Failed to find match for field 'noise'.
[graph_based_slam_node-5] ----------------------------
[graph_based_slam_node-5] searching Loop, num_submaps:2
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'noise'.

rsasaki0109, 2023-01-27:
Maybe slowing down the playback speed of rosbag would help.

jeychandar, 2023-01-27:
@rsasaki0109 I tried as you said, recorded the rosbag by slowing down the playback speed. Still same noise error

rsasaki0109, 2023-02-01:
I see. I'm a little too busy to respond right now...

jeychandar, 2023-02-01:
@rsasaki0109 Once you are free please let me know how to fix this error sir. Awaiting your reply.

rsasaki0109, 2023-02-04:
Is it possible to have your rosbag provided?

jeychandar, 2023-02-06:
@rsasaki0109 https://drive.google.com/file/d/1TBhBEjeQ_oIxmd-qDBHK1sQvTX6nY4YR/view?usp=sharing

rsasaki0109, 2023-02-11:
I can't find the ring when I converse your ros1 rosbag with rosbags.
Is the rosbag you provided correct?
```
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
[image_projection-4] Failed to find match for field 'ring'.
[image_projection-4] Failed to find match for field 'noise'.
```

jeychandar, 2023-02-11:
@rsasaki0109 Actually the rosbag that I sent to you is recorded with internal imu not external imu. I am able to solve the issue ring but noise unable to solve. Change image projection ouster struct uint_8t to uint_16t it will solve the ring issue.

rsasaki0109, 2023-02-12:
Have you calculated the following parameters?
https://github.com/rsasaki0109/li_slam_ros2/blob/develop/scanmatcher/param/lio.yaml#L9-L12
If not, please use the following package to find them.
https://github.com/gaowenliang/imu_utils

jeychandar, 2023-02-16:
@rsasaki0109 Ouster itself has some timestamp issues and they have mentioned in their community. And I don't have ouster lidar right now Sir.

rsasaki0109, 2023-02-16:
Can you share the link where the ouster timestamp issue is discussed?

jeychandar, 2023-05-03:
@rsasaki0109 Forgive me for the late reply stuck with another project. Timestamp issue has been corrected by ouster 2 weeks ago in their official repo. I converted pcap to rosbag. since the conversion contains /os_node/lidar_packets and /os_node/imu_packets topics, so I replayed the converted rosbag using their official repo **https://github.com/ouster-lidar/ouster-ros**   _roslaunch ouster_ros replay.launch  bag_file:=<path to rosbag file>  metadata:=<json file name>_ . When the recorded bag was playing in our li_slam_ros2 it gives ring and noise. ring has been solved using uint16_t in Image projection. Even though i reduce the playback speed it gives large velocity reset Imu preintegration. Could you please help me out with this issue. I have been dealing this issue for so long.

rsasaki0109, 2023-05-05:
@jeychandar 
I'm sorry, but I think I accidentally deleted the rosbag you provided. Could you please share it with me again? The link you gave me before has already expired.

jeychandar, 2023-05-05:
@rsasaki0109 I am sharing the pcap and metadata here sir. In case if i did wrong conversion can you check further. since rosbag has larger size I am unable to upload in mydrive.(Mydrive has limited storage). sorry for that.

metadata-- https://drive.google.com/file/d/17IAqYqTlZJi5qYtb_wAS9U32tok8r7Uj/view?usp=sharing
pcap-- https://drive.google.com/file/d/19ZNMHx5Ne9j9SgO36OejQmTnxzws3pmx/view?usp=sharing

**pcap to bag conversion link**: https://github.com/Krishtof-Korda/pcap-to-bag

**Note**: After converting the pcap to rosbag it contains only packet messages. You have to replay the converted rosbag into ouster replay and then record.

**ouster replay to record bag**: https://github.com/ouster-lidar/ouster-ros

**command line for reference**: roslaunch ouster_ros replay.launch      \
    bag_file:=<path to rosbag file>     \
    metadata:=<json file name>          # optional if bag file has /metadata topic

jeychandar, 2023-05-08:
@rsasaki0109 Hi sir I am managed to find out the issue and it is working properly. Our repo supports both internal and external IMU for ouster.The issue is based on sensor msg type format. Thanks for the support :)

closed 2023-05-08 by jeychandar

lidarmansiwon, 2023-07-12:
Hello, I am having the same problem as you.Currently using ouster lidar and external imu is using lord 3DM. Ring, noise problems are occurring the same as you. Please help. Which part did you modify and correct?

[image]
[image]

lidarmansiwon, 2023-07-12:
For your information, I changed the ring part to 16 based on what you said, so the ring problem disappeared, but the noise problem still occurs.

jeychandar, 2023-07-13:
@lidarmansiwon can you share rosbag info and also try to reduce the playback speed and see if that works.

lidarmansiwon, 2023-07-13:
https://balsam-feet-f83.notion.site/bag_backup-9fed659470f64498b27293a3df5df20c?pvs=4

bag file is so big. so I posted it in notion. so you can get bag in there! 
Now, I have ouster and lord 3DM imu. but when I get real sensor data, and using that for slam. it doesn't work too. I can see same error... please help me

lidarmansiwon, 2023-07-13:
[image]

Some modifications were made to the image_projection file during the debugging process. I changed the uint8_t ring part to 16 and just annotated the noise part. That's how I executed it, and the code didn't work anymore. When I echo the topic, I found that the data did not come out of the cloudHandler(), and after debugging, the problem occurred in the cloud_deskewed topic function.
There was a problem within the function, but the log "Waiting for IMU data ..." did not occur in the terminal.However, if (imuQueue.empty() || t_f > timeScanCur || t_b < timeScanNext) s were assigned to the false conditional statement, so the function did not spin. So let's process the comments as shown in the picture below
The projectPointCloud function started spinning and something started to be drawn on the rviz. That's the process I've done. I'm saying this because I'm wondering if it'll help you solve it. Thank you for trying to help.

[image]

CantLooseToAMoose, 2024-05-27:
@jeychandar 
> @rsasaki0109 I am sharing the pcap and metadata here sir. In case if i did wrong conversion can you check further. since rosbag has larger size I am unable to upload in mydrive.(Mydrive has limited storage). sorry for that.
> 
> metadata-- https://drive.google.com/file/d/17IAqYqTlZJi5qYtb_wAS9U32tok8r7Uj/view?usp=sharing pcap-- https://drive.google.com/file/d/19ZNMHx5Ne9j9SgO36OejQmTnxzws3pmx/view?usp=sharing
> 
> **pcap to bag conversion link**: https://github.com/Krishtof-Korda/pcap-to-bag
> 
> **Note**: After converting the pcap to rosbag it contains only packet messages. You have to replay the converted rosbag into ouster replay and then record.
> 
> **Pcap to rosbag converted rosbag link**: https://drive.google.com/file/d/1JFvBYXpI46_GKJKC9oYkweSCyG7eGGIE/view?usp=sharing You can replay and record in ouster_ros sir. may be this one will be helpful.
> 
> **ouster replay to record bag**: https://github.com/ouster-lidar/ouster-ros
> 
> **command line for reference**: roslaunch ouster_ros replay.launch bag_file:= metadata:= # optional if bag file has /metadata topic

Can you go into further details on how to convert the bag file to play the right messages? 
When i use `roslaunch ouster_ros replay.launch` only the original `/os_node/imu_packets` and `/os_node/lidar_packets` get played.
What topic would you expect to be played to?