rsasaki0109/li_slam_ros2 #23: Run bag met error gtsam::IndeterminantLinearSystemException

j953302, 2023-06-19:
Hello, I want to run li_slam with a V16s velodyne lidar and a 9 axis imu. I never change any default parameters in lio.yaml. 

My data is a indoor corridor loop. However, when li_slam run just a little time it occur below problem:
[image]

And this is my bag file.
[https://drive.google.com/drive/folders/1Ov5nnFHmqO3qDI_GLPtWwOS9HZtOQiP0?usp=sharing](url)

Could you help me figure it out? Thank you.

rsasaki0109, 2023-06-19:
I might be too busy this month to look at the rosbag. Could you please post a video of rviz or something similar? You might find the link below helpful:
https://github.com/TixiaoShan/LIO-SAM#prepare-imu-data

rsasaki0109, 2023-06-20:
@j953302 
I would also like to be provided with a yaml file.

rsasaki0109, 2023-06-20:
There may be a problem with imu timestamps.

> I had this problem too. In my case the reason was that some imu messages were sent with the same timestamp.
https://github.com/TixiaoShan/LIO-SAM/issues/165#issuecomment-1000247782

[image]
[image]

j953302, 2023-06-26:
@rsasaki0109 Hello, sorry too late to reply.
I have calculate the data. However, in #165 the problem is imu data has same timestamp. But my minimun value dt of  imu data is 0.000043201 s. It's not equal to 0. So I'm not sure #165's solution could help me.
Also, because LIO Sam suggest imu publish frequency up to 200hz. So my data is 200hz

j953302, 2023-06-27:
@rsasaki0109 Wow, it's really hepful. It fixed my first bug huuuuu...
Thank you.

closed 2023-06-27 by j953302

zhao-zhibo, 2023-12-09:
My problem was solved. It was a problem with the Eigen version. Changing the expression solved the problem.I asked a question on stackoverflow and someone else answered me.[stackoverflow](https://stackoverflow.com/questions/77627134/gtsam-error-report-assertion-sizeatcompiletime-dynamic-sizeatcompiletim)

szydgithub, 2024-04-23:
> My problem was solved. It was a problem with the Eigen version. Changing the expression solved the problem.I asked a question on stackoverflow and someone else answered me.[stackoverflow](https://stackoverflow.com/questions/77627134/gtsam-error-report-assertion-sizeatcompiletime-dynamic-sizeatcompiletim). But I don't understand why this can solve the problem. And I don’t understand why this problem occurs?

可以具体说一下是如何解决的吗？谢谢