rsasaki0109/li_slam_ros2 #25: Same data but not always successed. Jump & Large velocity sometimes.

j953302, 2023-06-27:
Hello @rsasaki0109 
Thanks for your soon reply. You have help me clean one bug in #23  . And I have add the check delta_t code in imu_preintegration.cpp.

However, when I keep play my bag data, I hope it could finish a loop.
But it will happen a lot of problem randomly.
Here I record some result here.
And I knew you are busy. But if you are free, please help me check my data or how could I avoid this randomly problem.
Thank you very much.

bag file:
[https://drive.google.com/drive/folders/1Ov5nnFHmqO3qDI_GLPtWwOS9HZtOQiP0?usp=sharing](https://github.com/rsasaki0109/li_slam_ros2/issues/url)

Result1:
First half everything is perfact. But other will jump out. And I will get Rviz2 info:
`[rviz2]: Message Filter dropping message: frame 'base_link' at time 1686970685.207 for reason 'Unknown'
`
[image]

Result2:
Same. But this time it jump early than result1. And show more info like:
`[imu_preintegration]: Large velocity, reset IMU-preintegration!
`
[image]

If I have other new found, I will update here. Thank you very much.

rsasaki0109, 2023-06-29:
@j953302 Can you provide the yaml file?

j953302, 2023-06-30:
@rsasaki0109 Sure. But I didn;t change anything.
[lio.zip](https://github.com/rsasaki0109/li_slam_ros2/files/11911438/lio.zip)
If you need more information, please tell me.

rsasaki0109, 2023-06-30:
@j953302 

The following output is being produced. Are you using image_projection (which is a node for scan distortion correction)? It can improve estimation accuracy if used.

```
[image_projection-4] Failed to find match for field 'time'.
[image_projection-4] [ERROR] [image_projection]: Point cloud is not in dense format, please remove NaN points first!
[image_projection-4] [WARN] [image_projection]: Point cloud timestamp not available, deskew function disabled, system will drift significantly!
[INFO] [image_projection-4]: process has finished cleanly [pid 1039676]
```

The accuracy will degrade if the following is not estimated with an actual IMU. For example, the following package could be useful: https://github.com/gaowenliang/imu_utils

```
      imuAccNoise: 3.9939570888238808e-03
      imuGyrNoise: 1.5636343949698187e-03
      imuAccBiasN: 6.4356659353532566e-05
      imuGyrBiasN: 3.5640318696367613e-05
```

rsasaki0109, 2023-06-30:
Also, setting ndt_resolution to 1.0 or slowing down the rosbag playback speed may improve the trajectory estimation.

rsasaki0109, 2023-06-30:
>  And I have add the check delta_t code in imu_preintegration.cpp.

By the way, it would be great if you could give me a PR for this.
If it's too much trouble, I can fix it myself.

rsasaki0109, 2023-06-30:
@j953302 
In my environment, there were no issues with localization even when playing the rosbag at 1.0x with your parameters, so I suspect the problem lies with your PC specs. I would recommend adjusting the ndt_num_threads or lowering the speed of rosbag playback. 
[image]

j953302, 2023-06-30:
> The following output is being produced. Are you using image_projection (which is a node for scan distortion correction)? Not using it can lead to a decrease in estimation accuracy.
> 
> ```
> [image_projection-4] Failed to find match for field 'time'.
> [image_projection-4] [ERROR] [image_projection]: Point cloud is not in dense format, please remove NaN points first!
> [image_projection-4] [WARN] [image_projection]: Point cloud timestamp not available, deskew function disabled, system will drift significantly!
> [INFO] [image_projection-4]: process has finished cleanly [pid 1039676]
> ```
> 
> The accuracy will degrade if the following is not estimated with an actual IMU. For example, the following package could be useful: https://github.com/gaowenliang/imu_utils
> 
> ```
>       imuAccNoise: 3.9939570888238808e-03
>       imuGyrNoise: 1.5636343949698187e-03
>       imuAccBiasN: 6.4356659353532566e-05
>       imuGyrBiasN: 3.5640318696367613e-05
> ```
[image_projection-4] Failed to find match for field 'time'. -> This problem will always occur but it will not affect my reconstruct. Is it ok?

And because imu_utilts is just for ros1. Maybe later I try to change it into ros2 and try it.

j953302, 2023-06-30:
> > And I have add the check delta_t code in imu_preintegration.cpp.
> 
> By the way, it would be great if you could give me a PR for this. If it's too much trouble, I can fix it myself.

It's my pleasure, but I didn't change too much code.
I change in image_projection.cpp. And add some code before it check if point cloud is dense cloud point. Like below:

      std::vector<int> indices;

      pcl::removeNaNFromPointCloud(*laserCloudIn, *laserCloudIn, indices);
      if (laserCloudIn->is_dense == false) {
        RCLCPP_ERROR(
          get_logger(), "Point cloud is not in dense format, please remove NaN points first!");
        rclcpp::shutdown();
      }

j953302, 2023-06-30:
Hello, it work stably after I change ndt_num_threads and rosbag play speed.
Here I put my result here. And I will keep optimate this result. Thank you very much.
[image]
[image]

rsasaki0109, 2023-07-03:
The issue seems to be resolved, so I'll close this issue.

closed 2023-07-03 by rsasaki0109