rsasaki0109/li_slam_ros2 #28: How to Check Mapping Accuracy

jeychandar, 2023-08-01:
@rsasaki0109 Hi Sir. Since you have working with mapping related field, I am curious to know about the mapping accuracy. Could you guide me how to check mapping accuracy sir. Thanks in advance.

rsasaki0109, 2023-08-01:
Is it something like comparing with RTK GNSS? I think the following link may be of reference.
https://github.com/autowarefoundation/autoware-documentation/pull/366

jeychandar, 2023-08-01:
We would like to check the 3D LiDAR mapping accuracy for indoor settings. Eventually the mapped scans will be compared against building information models (e.g. construction progress monitoring & verification).

rsasaki0109, 2023-08-01:
For indoor references, tracking is often done with total stations. Additionally, for map evaluation, comparisons can be made with floor plans or maps created with ground-based laser scanners.

jeychandar, 2023-08-01:
Okey sir will check it out

rsasaki0109, 2023-12-10:
This issue has been inactive for a while and will be closed for housekeeping. If it's still relevant, please feel free to reopen it with a new comment. Thank you for your contribution.

closed 2023-12-10 by rsasaki0109