rsasaki0109/li_slam_ros2 #35: Drfiting during the mapping process

wxyLuna2024, 2024-07-31:
Hello Sasaki San,
Thank you for integrating the IMU Integration from LIO-SAM to Lidar_slam_ros2. Using this repo, I created the best looking map with large scale dataset by far. The method is very reliable in ROS2 system and easy to use.However, I still have a couple of questions:
Some times weird drift happens. And eventually a small drift results in accumulative error that leads to failure of loop closure, especially for large scale dataset. For me the mapping was all perfect until the mapping hit a small drift.
[image]
[image]
Loop closure failing
Also another drift happens at this U-turn
[image]
[image]
Eventually resulted in misalignment in the lower left area, as well as aligning the map into a tilted plane.
[image]
The pointcloud plane were not even, and different sections are slightly tiled. Although I set imu input to be true and I see no error returned by IMUPreintegration, I still wonder if IMU input really comes into place.
[image]
I am using the default lio_bigloop.yaml set up here and any help will be greatly appreciated!!

rsasaki0109, 2024-07-31:
I think it's necessary to use a trimmed rosbag from the problematic area to visualize the imu_path and path in rviz, and to check the values before and after using the debug_flag to identify the cause.

If the accuracy of NDT is the issue, the parameters ndt_resolution, vg_size_for_input, and vg_size_for_map might need to be reduced by about half (though this requires modifying the source code, it might be better not to apply a voxel grid filter to the map).

wxyLuna2024, 2024-08-06:
Hello,

Thank you for your reply. The frequency of the IMU is 100 Hz. 

Also, as I am trying to evaluate the li_slam_ros2 performance with gt poses, I wonder how to evaluate its performance in a quantitative way, as the framework outputs trajectory in .g2o format, while most of the benchmark(like evo) uses pose matrices. I was referring to https://autowarefoundation.github.io/autoware-documentation/pr-366/how-to-guides/creating-maps-for-autoware/open-source-slam/compare-slam-algorithms/ but steps and tools were not stated clearly...

rsasaki0109, 2024-08-06:
One simple method might be to record the pose topic, and then convert it into a pose matrix using the following function:
https://github.com/rsasaki0109/li_slam_ros2/blob/develop/scanmatcher/src/scanmatcher_component.cpp#L493-L499

wxyLuna2024, 2024-08-07:
> One simple method might be to record the pose topic, and then convert it into a pose matrix using the following function: https://github.com/rsasaki0109/li_slam_ros2/blob/develop/scanmatcher/src/scanmatcher_component.cpp#L493-L499

Hello,
I was able to record the current_pose/ topic and use existing evo tool to do the evaluation. Thank you!